flxkrmr/eec-iv-reader-arduino #24: UART Message format array length number of bytes correct?

mercury64, 2023-02-17:
UART messages are 11 bytes long, why is the CART::startMessage 18 bytes?

flxkrmr, 2023-02-17:
Thanks for your interest in the project!

The start message differs from all other messages. I could not find anything about it in the docs, but used the infos in this project: https://github.com/babroval/ford-eec-iv-diagnostic/blob/master/DCL_communication_map.pdf (see blue message on the left). After sending this message the ECU goes into "DCL-communication-mode" and start sending the sync messages. What is weird, the start message is send with 2400 baud and the ECU starts sending sync messages with 9600 baud

flxkrmr, 2023-02-24:
Implemented in 452906b2bd2ff4d2686d64f7763628f636d0f63c

closed 2023-02-24 by flxkrmr