flxkrmr/eec-iv-reader-arduino #25: Need to implement parity checking for UART messages

mercury64, 2023-02-17:
As an example, the UART command to start CART is 0x4.
The module id is 0x1, the EEC node on the bus.
The last byte is the parity check byte.
Simply AND all the bytes together:
0x00...0x01, 0x04, 0x00, 0x00, 0x00 = 0x05

Command 0x04 go into CART mode.
Command 0x05 is use alternate calibration.
Command 0x80 is go permanently offline
Command 0x82 is echo module id

flxkrmr, 2023-02-17:
Do you mean the start message mentioned here https://github.com/flxkrmr/eec-iv-reader-arduino/issues/24?

These commands are interesting! Do you have more informations about the UART mode or some documents?

mercury64, 2023-02-18:
Yes, I have documents, as well as I have decompiled my EEC and have the code that runs on the EEC.