flxkrmr/eec-iv-reader-arduino #27: Allow ECUs that start with 19200 bauds

flxkrmr, 2023-02-17:


mercury64, 2023-02-18:
As far as I know, the initial UART message starts at 2400 baud for all.
The request is made to put in CART mode 0x04 using 2400 baud.
Then depending on the EEC strategy configuration, CART mode is at 19200 baud.
Once in CART mode, the baud can be changed 2400,  4800, 9600, or 19200.
Command code 0x81, DCL_DM_SetCommLinkBAUDRate                  # Set DCL baud rate
in CART mode changes the EEC baud rate.

flxkrmr, 2023-02-18:
The code currently only works with CART mode that starts with 9600 baud and is has trouble with units that start CART mode with 19200 baud. After CART initialisation it always switched baud rate to 2400 for all diagnostic communication. 

Do you know if all diagnostic commands work with all baud rates? I had trouble getting the error codes with 9600 baud, but I found nothing about this in the docs.

mercury64, 2023-02-18:
UART mode is only 2400.
You can only start CART mode in UART mode at 2400.
CART mode will start in whatever the EEC is programmed to.
CART mode can be changed once in this mode to 2400, 4800, 9600, or 19200.

I am unable to get any communication happening right now, probably a noisy circuit.

flxkrmr, 2023-02-19:
In my tests the communication seemed very resistant against noise or bad cable connections. Maybe check the delays between bytes and IWs. The EEC IV is more unforgiving with these

mercury64, 2023-02-20:
EEC-IV hardware code SME-105, software code ENVY once in cart mode starts in 9600.

What hardware code you have, and software code? 

I can decompile for you, and show the code in the EEC.
UART command code 0x04= GO CART MODE:
`5a6d: d7,19               jne   5a88             if (R33 = 4)  {
5a6f: 91,08,df            orb   Rdf,8            B3_Rdf = 1;
5a72: b1,02,32            ldb   R32,2            R32 = 2;`
R32 = 2 = 0x2H = 9600baud
# 00H                2400 baud
# 01H                4800 baud
# 02H                9600 baud
# 03H               19200 baud

flxkrmr, 2023-02-20:
How do I find the hardware- and software-code?

mercury64, 2023-02-20:
Software code, or calibration is the 3 or 4 digit bold uppercase letters and numbers located on the actual EEC label.
The hardware code is not bold, can start with SME-xxxx, SDxxxx, MAF-xxxx, MLP-xxx, etc.

Post a picture of your EEC-IV label.

flxkrmr, 2023-02-21:
My main testing ECU is VANE SME-401. This one starts in 9600 baud.
The ECU that starts in 19200 is BEAU SME-105.
And lastly one ECU that does not answer at all is 2GEA EST-230

mercury64, 2023-02-22:
The 2GEA EFI-SD230 must have the 81C62 chip.
More than likely it only has  81C61 chip. 
For DCL needs the second crystal that runs at 4.9152Mhz and the 81C62 CART chip.

I'll try and find the VANE and BEAU bins and will disassemble them with DCL UART/CART info. I will check in my own repository and will create a pull request.

mercury64, 2023-02-22:
Here is first commit of disassembled SME-401.
Have a read, I will update it more later.
It is fairly well commented.

Here are some links:
External Interupt from UART/CART serial data
https://github.com/mercury64/eec-iv-reader-arduino/blob/master/hardware/SME-401/SEAL_lst.txt#L5742

UART MODE
https://github.com/mercury64/eec-iv-reader-arduino/blob/master/hardware/SME-401/SEAL_lst.txt#L5754

CART MODE
https://github.com/mercury64/eec-iv-reader-arduino/blob/master/hardware/SME-401/SEAL_lst.txt#L5829

DEFINED EEC-IV DIAGNOSTIC MODES
https://github.com/mercury64/eec-iv-reader-arduino/blob/master/hardware/SME-401/SEAL_lst.txt#L6316

flxkrmr, 2023-02-22:
Damn, thats interesting. Thanks for this. Event though I'm not so good at reading assembler, I think I can pull some infos out of this

mercury64, 2023-02-22:
Comments are on the right hand side. I only commented the DCL stuff. I copied it from my EEC-IV SD48b KID2 v8 351w E4OD(Canadian). I've been disassembling it for about 3 years. The DCL code is near identical.
I'll push my documentation.