flxkrmr/eec-iv-reader-arduino #29: Continuous Memory Errors are deleted after KOEO run

flxkrmr, 2023-02-23:
It seems that all Errors in the Continuous Test Memory (Diagnostic mode 0x26) are delete after running the KOEO test (0x25).

@mercury64 Do you by any chance know something about this?

mercury64, 2023-02-24:
First thought is, that continuous memory codes are stored in Keep Alive Memory(KAM), yellow power wire on EEC.
If power is disconnected, the codes are gone.
Give me a few days, I finally have communication going with correct delays, CART mode working with UNO and software serial.
I'll test it over this weekend.

flxkrmr, 2023-02-24:
Power is not disconnected, so this should not be the problem.

I just found something in one of your documents, which might be the cause:
"If STI is ungrounded during the output of codes in Engine Off Test, all continuous
      self  test  codes are erased from KAM.  This allows for repair verification by the
      mechanic."

All tests where run on Ford Mondeo mk1 where STI is not connected at all

mercury64, 2023-02-24:
Yup, that would explain that.
Here is a link to more documentation:
https://drive.google.com/drive/folders/1UnA8qP_5oNhmdacBlSEwCUKwn0QFDHdr

closed 2023-02-24 by flxkrmr